TengerTechnologies/Fuzzyficator #1: Modification for BambuStudio

carribeiro, 2024-10-26:
Hi, I'm working on a project where having a fuzzy top is a customer wish. I'm testing it today and I've found a few issues, I opened this issue to talk about it first so I can understand the best way to proceed.

First of all, Bambu Studio generates "`; FEATURE Top surface`" lines not "`;TYPE:Top solid infill`" lines. I've changed that and the script is working (at least, processing lines as it should).

Now, I'm testing the result, but it seems to me that there's something strange. There's a lot of material at the end of each line. Seems that the material isn't  being evenly distributed on the segments as it should.

There are other artifacts - seems like the A1 Mini that I'm using don't like hitting the higher spots when traveling and go home, which is the Bambulab behavior to correct misses steps I think. But we can fix it later - for now I'm focusing on the even material distribution.

Maybe we can discuss theses fixes here, instead of opening separate issues, that would be a bit faster at the moment.

carribeiro, 2024-10-26:
One such example:

```
2024-10-26 15:58:36,210 - DEBUG - Extracted coordinates: {'X': 90.333, 'Y': 44.517, 'E': 0.00268}
2024-10-26 15:58:36,210 - DEBUG - Current point: (90.333, 44.517, 8.08), Total extrusion: 0.00268
2024-10-26 15:58:36,210 - DEBUG - Calculated distance between (81.458, 47.292, 8.08) and (90.333, 44.517, 8.08): 9.29872303061017
2024-10-26 15:58:36,210 - DEBUG - Interpolating between (81.458, 47.292, 8.08) and (90.333, 44.517, 8.08) with 18 segments
2024-10-26 15:58:36,210 - DEBUG - Generated interpolated point: (81.458, 47.292, 8.08, 0.0001488888888888889)
2024-10-26 15:58:36,211 - DEBUG - Generated interpolated point: (81.95105555555556, 47.13783333333333, 8.099415488938726, 0.0001488888888888889)
2024-10-26 15:58:36,211 - DEBUG - Generated interpolated point: (82.44411111111111, 46.98366666666667, 8.100281614310358, 0.0001488888888888889)
2024-10-26 15:58:36,211 - DEBUG - Generated interpolated point: (82.93716666666667, 46.8295, 8.084062652683764, 0.0001488888888888889)
2024-10-26 15:58:36,211 - DEBUG - Generated interpolated point: (83.43022222222223, 46.675333333333334, 8.111982774515605, 0.0001488888888888889)
2024-10-26 15:58:36,211 - DEBUG - Generated interpolated point: (83.92327777777777, 46.521166666666666, 8.091217044205507, 0.0001488888888888889)
2024-10-26 15:58:36,211 - DEBUG - Generated interpolated point: (84.41633333333333, 46.367000000000004, 8.101582480151492, 0.0001488888888888889)
2024-10-26 15:58:36,211 - DEBUG - Generated interpolated point: (84.90938888888888, 46.212833333333336, 8.11176886941478, 0.0001488888888888889)
2024-10-26 15:58:36,211 - DEBUG - Generated interpolated point: (85.40244444444444, 46.05866666666667, 8.102876134423708, 0.0001488888888888889)
2024-10-26 15:58:36,223 - DEBUG - Generated interpolated point: (85.8955, 45.9045, 8.09642839914839, 0.0001488888888888889)
2024-10-26 15:58:36,223 - DEBUG - Generated interpolated point: (86.38855555555556, 45.75033333333334, 8.12740495651404, 0.0001488888888888889)
2024-10-26 15:58:36,223 - DEBUG - Generated interpolated point: (86.88161111111111, 45.59616666666667, 8.106718795720306, 0.0001488888888888889)
2024-10-26 15:58:36,223 - DEBUG - Generated interpolated point: (87.37466666666667, 45.442, 8.11762321639275, 0.0001488888888888889)
2024-10-26 15:58:36,224 - DEBUG - Generated interpolated point: (87.86772222222223, 45.28783333333334, 8.117462188675805, 0.0001488888888888889)
2024-10-26 15:58:36,224 - DEBUG - Generated interpolated point: (88.36077777777777, 45.13366666666667, 8.081916284547605, 0.0001488888888888889)
2024-10-26 15:58:36,224 - DEBUG - Generated interpolated point: (88.85383333333333, 44.9795, 8.081240098034385, 0.0001488888888888889)
2024-10-26 15:58:36,224 - DEBUG - Generated interpolated point: (89.34688888888888, 44.82533333333333, 8.110804080028492, 0.0001488888888888889)
2024-10-26 15:58:36,224 - DEBUG - Generated interpolated point: (89.83994444444444, 44.67116666666667, 8.11989644794261, 0.0001488888888888889)
2024-10-26 15:58:36,224 - DEBUG - Generated interpolated point: (90.333, 44.517, 8.105644183060308, 0.0001488888888888889)
2024-10-26 15:58:36,224 - DEBUG - Added interpolated G1 command: G1 X81.4580 Y47.2920 Z8.0800 E0.0001
2024-10-26 15:58:36,225 - DEBUG - Added interpolated G1 command: G1 X81.9511 Y47.1378 Z8.0994 E0.0001
2024-10-26 15:58:36,225 - DEBUG - Added interpolated G1 command: G1 X82.4441 Y46.9837 Z8.1003 E0.0001
2024-10-26 15:58:36,225 - DEBUG - Added interpolated G1 command: G1 X82.9372 Y46.8295 Z8.0841 E0.0001
2024-10-26 15:58:36,225 - DEBUG - Added interpolated G1 command: G1 X83.4302 Y46.6753 Z8.1120 E0.0001
2024-10-26 15:58:36,225 - DEBUG - Added interpolated G1 command: G1 X83.9233 Y46.5212 Z8.0912 E0.0001
2024-10-26 15:58:36,225 - DEBUG - Added interpolated G1 command: G1 X84.4163 Y46.3670 Z8.1016 E0.0001
2024-10-26 15:58:36,225 - DEBUG - Added interpolated G1 command: G1 X84.9094 Y46.2128 Z8.1118 E0.0001
2024-10-26 15:58:36,225 - DEBUG - Added interpolated G1 command: G1 X85.4024 Y46.0587 Z8.1029 E0.0001
2024-10-26 15:58:36,225 - DEBUG - Added interpolated G1 command: G1 X85.8955 Y45.9045 Z8.0964 E0.0001
2024-10-26 15:58:36,226 - DEBUG - Added interpolated G1 command: G1 X86.3886 Y45.7503 Z8.1274 E0.0001
2024-10-26 15:58:36,226 - DEBUG - Added interpolated G1 command: G1 X86.8816 Y45.5962 Z8.1067 E0.0001
2024-10-26 15:58:36,226 - DEBUG - Added interpolated G1 command: G1 X87.3747 Y45.4420 Z8.1176 E0.0001
2024-10-26 15:58:36,226 - DEBUG - Added interpolated G1 command: G1 X87.8677 Y45.2878 Z8.1175 E0.0001
2024-10-26 15:58:36,226 - DEBUG - Added interpolated G1 command: G1 X88.3608 Y45.1337 Z8.0819 E0.0001
2024-10-26 15:58:36,226 - DEBUG - Added interpolated G1 command: G1 X88.8538 Y44.9795 Z8.0812 E0.0001
2024-10-26 15:58:36,226 - DEBUG - Added interpolated G1 command: G1 X89.3469 Y44.8253 Z8.1108 E0.0001
2024-10-26 15:58:36,226 - DEBUG - Added interpolated G1 command: G1 X89.8399 Y44.6712 Z8.1199 E0.0001
2024-10-26 15:58:36,226 - DEBUG - Added interpolated G1 command: G1 X90.3330 Y44.5170 Z8.1056 E0.0001
2024-10-26 15:58:36,227 - DEBUG - Added original G1 command: G1 X90.333 Y44.517 E.00268
```

Note that the total extrusion was of 0.00268; but it created several 0.0001 segments and later put all material on the last command. I guess this isn't the intended behavior, so I'm asking if you have the correct version to test!

TengerTechnologies, 2024-10-27:
The script is only tested with Marlinflavour gcode generted in Prusaslice. Thats why it doesn't work. I'm currently working on implementations for Orcaslicer and Cura. Bambuslicer is on my list to. I hope that I have the Orcaslicer version ready by the end of the day. So you could use that one for your project. (Sorry that it is a bit inconvenient, but you can use Orcaslicer for Bambu printers). 
As for the nozzle colliding problem, I suggest increasing the Z-hop value.

carribeiro, 2024-10-28:
Hi, it's ok, I know it's not supposed to be used with Bambu right now, that's why I'm working on it. Thre reason I asked you is because I found it strange that  the cde is using such small segments with `E0.0001` and then extruding all the filament on the last segment. Doesn't seem right to me and I thought you must have a update version, but I'll work with the one I got here on GitHub.

carribeiro, 2024-10-28:
Just to make it clear:

```
2024-10-26 15:58:36,210 - DEBUG - Extracted coordinates: {'X': 90.333, 'Y': 44.517, 'E': 0.00268}
```

Its a 9.298mm long line with 0.00268 extrusion that's split into 18 segments. So I assume that it should extrude 0.0001488 at each segment as to add to a total of 0.00268. But it does 17 segments of `E0.0001` followed by a full `E.00268` at the end.

carribeiro, 2024-10-29:
Nevermind the previous comment. The actual problem is a bit different: the last segment shouldn't be inserted after the interpolated segments as the code already goes to the last point, and that's why I was seeing some excess material building up at the end of the lines.

TengerTechnologies, 2024-10-29:
> Nevermind the previous comment. The actual problem is a bit different: the last segment shouldn't be inserted after the interpolated segments as the code already goes to the last point, and that's why I was seeing some excess material building up at the end of the lines.

Thank you, I've seen that. It is a bug from testing, the original G1 command should be inserted as comment.

carribeiro, 2024-10-30:
As for the adaptation for Bambu, a few notes.

BambuStudio doesn't generate conventional GCODE files. Instead it generates a 3mf files that has a LOT more information. But the 3mf file is actually just a zipper container and the GCODE file is inside there.

The instructions are here:

https://forum.bambulab.com/t/altering-g-code/72479

Another idea that is applicable to other slicers is to use Hilbert as the top surface. It creates a pattern that is already interesting for effects such as fuzzy.