TengerTechnologies/Fuzzyficator #2: Can't get it to work with Cura gcode.

doctormord, 2024-10-29:
First of all thanks for your efforts. :-)

I tried your script with a Cura generated GCode file but it does not alter the top layer. I use a simple cube for the test.

[image]
[image]
[image]

Any hints?

Best regards

Christian

doctormord, 2024-10-29:
Ah, Cura isn't clearly comment top layer skin but "SKIN".

[input.zip](https://github.com/user-attachments/files/17558578/input.zip)

doctormord, 2024-10-29:
Preparing the file with ORCASLICER does work if i change the .py

from 

`if line.startswith(';TYPE:Top solid infill'):`

to

`if line.startswith(';TYPE:Top surface'):`

TengerTechnologies, 2024-10-29:
Thank you for your issue documentation. The script is only tested in Prusaslicer. You will have to be a bit patient but I'm working on a cura version and it will eventually be done. I'm aware of the changes for Orcaslicer and the Orca version of the script is nearly done. I just have to test it before publishing. The cura version is a bit more difficult.