TengerTechnologies/Fuzzyficator #4: Wrong movements when having holes in top surface

TimoRaab, 2024-11-09:
When having holes in the top surface the script handles some movements wrong, I'm still investigating why the problem occurs. 
The first image is without the script, the second with the script. Here you can see some wrong extrusion movements. They are also present in the print, so not only a visual problem. 

I also included the stl and the gcode files associated with it. I will look further into it and update it.
[image]
[image]

[Cube_Fuzzy.zip](https://github.com/user-attachments/files/17686715/Cube_Fuzzy.zip)

FelipeRenault, 2025-01-23:
Any luck? I might have a look as well, just to see if I can find anything

jranger615, 2025-02-26:
seeing this as well

jranger615, 2025-02-26:
Looks like TimoRaab fixed this in Fuzzyficator_Orcaslicer.py by setting the below two lines:
170: elif not in_top_solid_infill and 'G1' in line and 'Z' in line: # only change height when it is not doing the top surface
206: elif in_top_solid_infill and line.startswith('G1') and ('X' in line or 'Y' in line or 'Z' in line):

This doesnt fix the new code but it fixed the Archived Orcaslicer version

michalkomanec, 2025-03-21:
Tried it in the current code and it fixed the problem in my print!

updated code lines:
286: elif not self.in_top_solid_infill and 'G1' in line and 'Z' in line:
288: elif (self.in_top_solid_infill or (self.in_bridge and self.config.lower_surface)) and line.startswith('G1') and ('X' in line or 'Z' in line):

Thank you for mentioning it.