TengerTechnologies/Fuzzyficator #10: Doesn't work with absolute extrusion mode

Kato0909, 2024-11-14:
Hello!

I am using PrusaSlicer and old i3 clone and have a problem - the extruded amount doesn't calculate correctly.
I use absolute extruder position in G-Code, so at the end of a layer the value of E in command can reach several tenth of mm, for example: (all examples from the same 2cm cube sliced with same settings except Fuzzyficator script on or off)
```
G1 X93.401 Y91.159 E31.26267
G1 X93.401 Y91.159 F7200
G1 X92.895 Y91.159
G1 F1500
G1 X91.159 Y92.895 E31.32844
G1 X91.159 Y92.895 F7200
G1 X91.159 Y92.389
G1 F1500
G1 X92.389 Y91.159 E31.37504
G1 X92.389 Y91.159 F7200
G1 X91.882 Y91.159
G1 F1500
G1 X91.159 Y91.882 E31.40243
G1 E30.40243 F2400
```

after script I got the next g-code

```
; G1 X93.401 Y91.159 E31.26267
G1 X93.401 Y91.159 F7200
G1 X92.895 Y91.159
G1 F1500
G1 X93.4010 Y91.1590 Z20.0000 E4.4755
G1 X93.0807 Y91.4070 Z20.1546 E4.7979
G1 X92.7604 Y91.6550 Z20.0676 E4.5390
G1 X92.4401 Y91.9030 Z20.1666 E4.8482
G1 X92.1199 Y92.1510 Z20.0951 E4.6003
G1 X91.7996 Y92.3990 Z20.0049 E4.4758
G1 X91.4793 Y92.6470 Z20.1319 E4.7126
G1 X91.1590 Y92.8950 Z20.0000 E4.4755
; G1 X91.159 Y92.895 E31.32844
G1 X91.159 Y92.895 F7200
G1 X91.159 Y92.389
G1 F1500
G1 X91.1590 Y92.8950 Z20.0000 E6.2750
G1 X91.4050 Y92.5478 Z20.0607 E6.3467
G1 X91.6510 Y92.2006 Z20.0086 E6.2765
G1 X91.8970 Y91.8534 Z20.1043 E6.4849
G1 X92.1430 Y91.5062 Z20.1523 E6.7145
G1 X92.3890 Y91.1590 Z20.0000 E6.2750
; G1 X92.389 Y91.159 E31.37504
G1 X92.389 Y91.159 F7200
G1 X91.882 Y91.159
G1 F1500
G1 X92.3890 Y91.1590 Z20.0000 E10.4675
G1 X91.9790 Y91.4000 Z20.0328 E10.5026
G1 X91.5690 Y91.6410 Z20.1343 E11.0419
G1 X91.1590 Y91.8820 Z20.0000 E10.4675
; G1 X91.159 Y91.882 E31.40243
G1 E30.40243 F2400
G92 E0
```

The problem is in generated E values. They should be in the same range as values generated by slicer (around 31mm in this example), but they are all over the place (4, 6, 10mm, the longer XY travel, the smaller E values).

I checked the code and found an issue in interpolate_with_constant_resolution() function, which generates interpolated points. 

The X and Y coordinates are calculated accurately, its something like this:
X<sub>i</sub> = X<sub>start</sub> + (X<sub>delta</sub> *  i / n<sub>segments</sub>) , where `i` is the number of current interpolated point. 
Basically the stating point of a movement is saved and the only thing that being interpolated is delta between start and end of a movement.

However, when function calculates E value, it interpolates the whole E value, not a delta between start and end of E, something like this
E<sub>i</sub> = E<sub>start</sub> *  i / n<sub>segments</sub>
The result is that instead of extruding 30mm on filament on top layer, the code barely extrudes anything on top layer.

This function will be working correctly if before every move the slicer resets extruder position (i.e. G92 E0 command), but it doesn't work with absolute positioning. I couldn't manage to get slicer and printer to cooperate in relative extrusion mode.

Here's the screenshot of a preview of g-code in the middle of the 2cm cube, the extruded amount between start and end of a move is 0.00mm, basically the only variance in E value is a z_displacement_compensation, while original G-code extruded 0.7mm
[image]

The thing I suggest to implement is E calculation similar to X calculation.
E<sub>i</sub> = E<sub>start</sub> +   ( E<sub>delta</sub> *  i / n<sub>segments</sub> ) * compensation_factor
This calculation will work in both relative and absolute mode (in relative mode E<sub>start</sub> is equal to zero)

This need some rework of interpolate_with_constant_resolution(), and the way original E value is transferred into function (not as a number, but as part of start and end points). I think I'll do that sometime later on my end, but I don't know how to use git to transfer these changes to you.

Also, on line 63-64 the function saves "compensated_extrusion" value, but logging shows uncompensated value of "extrusion_per_segment".

Sorry if explanation is hard to follow, English isn't my native language.

turtle-bazon, 2024-12-16:
Same for me. Was need to examine code to understand why script is not working. I think that one line in readme could solve this.

bsafh, 2024-12-23:
> Same for me. Was need to examine code to understand why script is not working. I think that one line in readme could solve this.

Not for printers insisting on absolute extrusion, like my old but reliable Geeetech A30 - no way to get it working in this case.

MrMossevig, 2025-03-28:
Same here. Spent some time debugging it and I could just switch when I found it. I would just add a **3D-printer Requirements** in the README.