TengerTechnologies/Fuzzyficator #16: Post-Processing Script "..." on file "...gcode" failed. Error code: 9009

adho6298, 2024-12-24:
I am getting this issue in both bambu slicer and orca slicer. Not sure how to proceed. 

[image]

johann-miller, 2024-12-26:
I have the same issue in PrusaSlicer. Investigating it a little now.

adho6298, 2024-12-26:
I believe error code 9009 is python specific. But it has to do with python not being found or something. I am not a python dev despite having used python a handful of times. I'm not good at troubleshooting it.

meTeaDee, 2024-12-28:
I have the same issue in OrcaSlicer 2.2.0 and Bambu Studio 1.10.1.50 .

adho6298, 2025-01-05:
[image]
Just for reference, I recently reinstalled windows, so this is the same error on a fresh install of bambu studio, and python. 
Here is my input into my post processing script, and the error I am getting when I click "slice plate"

adho6298, 2025-01-05:
Just tried installing prusa slicer to see if I could replicate exactly what the video showed and I got the same error again.

[image]

adho6298, 2025-01-05:
I SOLVED IT!

error 9009 is a python error that means "bad path"
So i looked at my paths and assumed one of them was bad. Starting with the first one that leads to python.

After installing python and running "where python" in the command window, it gives a path, BUT that path is actually not good (yet).

How I fixed:
> Open a new file explorer 
> Navigate to the path produced by "where python". In my case it was in windows apps. There I found the python application. 
> Click on python and it will open a windows app page to python that says "get"
> get
I did this for the python3 file as well. 

After that it worked in bambu slicer.

adho6298, 2025-01-05:
Closing because I forgot

closed 2025-01-05 by adho6298