TengerTechnologies/Fuzzyficator #18: Orca Slicer Error code: 1

Sacrasang, 2025-01-03:
Orca Slicer error text:

Post-processing script
"C:\Users\David.DAVIDSPC\AppData\Local\Programs\Python\Python313\python.exe"
"C:\Program Files\Orcaslicer\Fuzzyficator.py" on file
C:\Users\DAVID~1.DAV\AppData\Local\Temp/orcaslicer_model/Wed_Jan_01/13_53_15#42756#3
70/Metadata\.42756.1.gcode.pp failed.
Error code: 1

I have confirmed my install of python is operational. Also opened the fuzzyficator.py script which is the latest release version.

TengerTechnologies, 2025-01-03:
@Sacrasang Can you please check the log at C:/users/yourUser/fuzzy_script.log ?

Sacrasang, 2025-01-03:
> @Sacrasang Can you please check the log at C:/users/yourUser/fuzzy_script.log ?

No log file is being generated. Or at least none I can find with a search.

markflay, 2025-01-09:
A quick review showed that PrusaSlicer is encoding as binary gcode (and I'm guessing others do too). Turning this off allowed it to complete.

Makemo07, 2025-01-13:
@markflay How do I change it in Prusa Slicer? Could not find the setting in Orca.

Have the error message as @Sacrasang:

Post-processing script "C:\Users\XXX\AppData\Local\Programs\Python\Python310\python.exe"
"C:\XXX\Fuzzy skin\Fuzzyficator.py" -run 1
-zMin 0 -zMax 0.5 -resolution 0.3 -ConnectWalls 1 -compensateExtrusion 1 on file
C:\Users\XXX\AppData\Local\Temp/orcaslicer_model/Mon_Jan_13/12_53_45#23744#49/Metadata\
.23744.0.gcode failed.
Error code: 1

and no error log can be found. Tried it with Python 3.13 on another computer - same error.

markflay, 2025-01-13:
In PrusaSlicer it's a setting under Printers -> General -> Firmware --> "Supports binary G-code"

Reading the literature, the main difference between turning this on / off is the file format it produces. With it on the file size is much smaller, with it off it is roughly 4x larger but the fuzzyficator script doesn't seem to know how to decode the bgcode file that is produced with it set to true.

In my slicer I have saved a second profile so I can have one with this flag off if I want a fuzzy surface and one with it on if I want standard. 

I am relatively new to 3D printing so I haven't used Orca slicer at all yet.

Makemo07, 2025-01-18:
Ok, found the error. I did not use the right script.

In my case (Bambu Lab X1C + Orca) "Fuzzyficator_Bambustudio.py" has to be used.

Landhus, 2025-01-24:
I was having the same error on my Ender 3V2, Error code 1, but then after I added " -run 1", Orca Slicer then saves the GCODE file, but when I open the file on Orca Slicer to see if it worked, there is no fuzzyskin on the top layer.

If I keep FuzzySkin off, there is no fuzzyskin on the part, but when I leave it on, only where Orca Slicer shows fuzzyskin on preview is where it ends up on the gcode file.

I'm kinda lost on how to proceed.

Landhus, 2025-01-27:
Just to add to my previous comment, changed the command to ### "C:\Users\%%%% %%%\AppData\Local\Microsoft\WindowsApps\python.exe" "C:\Users\%%%% %%%\Documents\Fuzzyficator\Fuzzyficator.py" -zMin 0 -zMax 0.5 -resolution 0.3 -connectWalls 1 -topSurface 1 -compensateExtrusion 1 -run 1;### 

And it worked. On PrusaSlicer. Same command does nothing on Orca Slicer. The log on Orca says: 
2025-01-24 15:33:31,131 - INFO - Script started
2025-01-24 15:33:31,220 - DEBUG - Support contact distance: 0.2
2025-01-24 15:33:31,431 - INFO - Fuzzy skin G-code generation completed successfully.

After this, the GCODE file has no fuzzy skin. But when executed on Prusa Slicer, it works.

dARoB2108, 2025-03-20:
I struggled for 3 days with this issue and the resolution was quite simple but a mistake I didn't even know it existed.

For downloading of the script, it is NOT ok to do right mouse click and "Save file as", as I did.
You need to click on the script, and then select "download raw file".

[image]

This finally works now!

tsem006, 2025-03-22:
same problem error code 2, tried everything but it doesn't work

[image]