TengerTechnologies/Fuzzyficator #19: Missing lines when trying to fuzzy surface around a circle/hole

adho6298, 2025-01-05:
[image]

As you can see, the script is missing lines on the top layer, and extruding over the hole when it should be traveling.

schnoesel666, 2025-02-14:
With the current versions (Orca Slicer 2.2.0, Python 13.13, Windows 10 with current updates (2024-01-11) it's the same for me. Tangential to circles/curves there's one line that's prominent. Maybe it has something to do with the wall connection setting?

jranger615, 2025-02-26:
can confirm, i am seeing this as well. Ive changed a bunch of settings, resolutions, etc and this percists. It doesn't matter which was you put the object either, it consistently in the same locations

praetp, 2025-06-15:
Yes, I confirm this problem. Seen with Bambu Studio 2.1.0.