TengerTechnologies/Fuzzyficator #20: Orca Slicer Error Code 9009

schnoesel666, 2025-01-11:
When slicing the following error code appears:

```
Post-processing script "C:\Users\***\AppData\Local\Microsoft\WindowsApps\python.exe"
"C:\Users\***\Downloads\Fuzzyficator.py" on file
C:\Users\***\AppData\Local\Temp/orcaslicer_model/Sat_Jan_11/12_13_12#32056#57/Metadat
a\.32056.0.gcode failed.
Error code: 9009
```

Orca Slicer 2.2.0
Python 13.13
Windows 10 with current updates (2024-01-11)

I tried it with and without admin rights without success. I also tried changing it to python3.exe with the same error code.

jilleb, 2025-02-06:
I had this same error, but... after changing the path from the Local/Microsofto folder to C:\Users\****\AppData\Local\Programs\Python\Python311-32\python.exe

No more errors!

willsy32, 2025-02-08:
Thank you Jileb! This should be the default file path