TengerTechnologies/Fuzzyficator #29: No fuzz in gcode

master-EH, 2025-04-28:
I have followed the complete step-by-step tutorial and resolved all errors I had using other people's issues, but there is no fuzz showing in the gcode viewer.

[image]

[image]

I have no idea what I'm doing wrong.

littlealiencheese, 2025-09-16:
@master-EH 

It started working for me after I changed the "Absolute Extrusion Mode (M82)" setting.
The script exits immediately if it detects M82.
To fix this:
- Go to Printer Settings > General > Relative Extruder Coordinates and enable the checkbox.
- In Printer Settings > Custom G-code > G-code after layer change, add the line: G92 E0.
I’m not sure how this affects other settings, but at least I managed to get it working

littlealiencheese, 2025-09-16:
Also the approximation of lines in G2 and G3 conflicts.

littlealiencheese, 2025-09-16:
Ah, it seems you have a completely different story. You need to add -resolution and set the value to something less than 0.1, for example. Your circles have very short segments, and the script simply doesn't work on them.

TengerTechnologies, 2025-09-16:
@master-EH 

In your screenshot you wrote ;FurryTool instead of ;FuzzyTool